nulllaborg/arduino_nulllab #6: In Arduino IDE cannot install board manager

Pat64000, 2021-12-29:
In Arduino IDE, after using the additional boards manager "https://raw.githubusercontent.com/nulllaborg/arduino_nulllab/master/package_nulllab_boards_index.json",
this is the output content when trying to install "Nulllab AVR Compatible Boards v1.0.2" -->

Downloading packages
nulllab avr compatible boards:avr@1.0.2
Installing platform nulllab avr compatible boards:avr@1.0.2
Failed to install platform: nulllab avr compatible boards:avr.
Error: 13 INTERNAL: Cannot install platform: installing platform nulllab avr compatible boards:avr@1.0.2: testing local archive integrity: testing archive size: fetched archive size differs from size specified in index

 - OS: Windows 11 
 - Arduino IDE v 2.0.0-rc2

nulllaborg, 2021-12-30:
please install 2.0.0  it is the laster version

Pat64000, 2021-12-30:
It's OK now with 2.0.0 version.
Thank you.

closed 2021-12-30 by Pat64000