nulllaborg/arduino_nulllab #12: eeprom只能读取无法写入

Hoshino-itsuki, 2022-02-22:
无论选择uno版型还是DIY模式选择1K eeprom，当读取eeprom的时候只能读到0-299的地址，EEPROM.length()返回的长度只有0-299。
[image]
[image]
当试图写入到一个地址的时候，程序将会卡在写入的那个地方，无法向下执行！
[image]
我尝试过核心库中的eeprom_write_byte和eeprom_write32也是同样的问题。
[image]
[image]

nulllaborg, 2022-03-04:
使用的外部晶振 有此问题，但是内部晶振 没问题

nulllaborg, 2022-03-05:
经过分析外部晶振时，setup 里面 write操作正常，但是 到loop里面 就会卡死

nulllaborg, 2022-03-06:
切换成外部晶振的时候 把 内部晶振32M disable。导致操作flash失败
https://github.com/nulllaborg/arduino_nulllab/commit/8d7688b0062e1c6e0600c56e27d2d8d63953b1b1

closed 2022-03-06 by nulllaborg