nulllaborg/arduino_nulllab #18: ISP Pins do not match AVR328P

jashsheth, 2022-07-17:
Hi Community

On referring the LQF8 Nano schematic https://github.com/nulllaborg/arduino_nulllab/blob/master/doc/LGT328P-LQFP32-Nano.pdf
I see the pins for ISP programming not matching the ones on the original 328P

Please let me know the wiring connections as well as the software settings needed to be done to upload firmware via USBASP which has MISO,MOSI,SCK and RST pins for programming

Help is much appreciated

blazej222, 2022-07-19:
I don't think it's possible to use USBASP to do the job here. I'm sure you can use another Arduino as ISP programmer for the LQF8 Nano, as described [here](https://github.com/dbuezas/lgt8fx/blob/master/lgt8f/libraries/LarduinoISP/readme.md) or [here](https://github.com/brother-yan/LGTISP).

The second repo contains source code and hex file on how to make the conversion.

nulllaborg, 2022-07-21:
> Hi Community
>
> On referring the LQF8 Nano schematic https://github.com/nulllaborg/arduino_nulllab/blob/master/doc/LGT328P-LQFP32-Nano.pdf I see the pins for ISP programming not matching the ones on [...]

Atmel328P usb spi download program, but LGT8F328P use swd donload program

closed 2023-02-20 by nulllaborg