nulllaborg/arduino_nulllab #27: Error updating from 1.0.3 to 2.0.0.  in IDE 2.3.6 fot Windows

AlexanderMKochkin, 2025-04-19:
**Describe the bug**
Failed to install platform: 'Nulllab_AVR_Compatible_Boards:2.0.0'. 13 INTERNAL: Failed to install platform: installing platform nulllab avr compatible boards:avr@2.0.0: checking local archive integrity: checking archive checksum: missing checksum for: master.zip

Ardiunщ IDE 2.3.6

**To Reproduce**
Steps to reproduce the behavior:
1. follow installation/update instructions
2. See error

**Screenshots**
https://disk.yandex.com/d/St58unCOt0BHgg

**Desktop :**
 - OS: Windiws 11 HOME
 - Browser : chrome
 - Version: Arduino IDE 2.3.6

Intechgreater, 2026-02-26:
Same for me in IDE 2.3.8

nulljun, 2026-09-20:
fix commit： 9208e3f66e4a19144fb05dda1bb84151ae320b44

closed 2026-09-20 by nulljun